krycha88/GUI-Generic #66: PZEM pomiar na OLED

KlimasTech, 2022-07-31:
Fajnie było by mieć informacje o aktualnym napięciu i natężeniu oraz mocy na wyświetlaczu OLED

Hrumque, 2022-08-01:
Jeśli problemem jest ogarnięcie "graficzności" tylu pomiarów (i ich wizualizacji na kilku różnych LCD/OLEDach o różnych rozdzielczościach) - to już wspominałem chyba - jakby była opcja włączenia UARTu na soft-serial jakiś na jednym GPIO wypluwać komplet pomiarów wszystkich i stanów, wg jakiegoś prostego a jawnego protokołu - to kto chętny na wyświetlacze itp, to sobie na kolanie dopisze własną obsługę dowolnego wyświetlacza na swoim ulubionym procku i w ulubionym formacie. 

Wystarczy, by każda linijka transmisji zaczynała się od np znaku/nr , by rozdzielić ew. wielofazowe pomiary, czy kilka czujników temperatury, i gotowe (gdzie xx - nr kanału pomiarowego, czujnika itp - na wszelki wypadek dwie cyfry, jak kiedyś ktoś wiecej niż 9 czujników wsadzi - 8 DSów, jakieś BMP280 i coś jeszcze, i byłby problem) 
Txx temperatury
Pxx cisnienia
Hxx wilgotnosci
Lxx luxy 
Axx ampery
Wxx waty
Vxx wolty
Xxx VAry 
Kxx kWh 
itd.  i po każdej takiej linijce znak sterujący nowej linii /n i gotowe. Po stronie dowolnego AVR wydłubanie takich stringów z bufora to banał
np. 
T01:21.75/n
P01:1024/n
i w kółko, co wykonany pomiar - wysłanie wyników. Albo co sekunde, albo jak tam wygodnie w sofcie to zaszyć

krycha88, 2022-08-03:
Dodałem takie wsparcie dla OLEDa.
https://github.com/krycha88/GUI-Generic/commit/ef094692f4bf9e7420366dc47fe53e7e579fa44d

closed 2022-08-05 by krycha88