krycha88/GUI-Generic #67: Konfig LED - wybór osobnego GPIO lub innego leda/ledów już zajętych

Hrumque, 2022-08-01:
Konfig LED (czy inaczej "status led" który:
- miga powoli gdy brak połączenia wifi/cloud
- miga szybko w trybie konfiguracji na 192.168.4.1
- świeci ciągle w całej pozostałej historii życia modułu ;) 
można by było zutylizować to mądrzej.
Gdyby tak w konfigu było można ustawiać  Konfig LED: WYŁACZONY/GPIOx/ i trzecia opcja: **Użyj led/ledów od przekaźników** 

W tym 3cim trybie flaga "zapal status led/zgaś status led" po prostu była XOR dla każdego GPIO/leda przypisanego od przekaźników. 

W normalnym trybie pracy każdy z ledów (lub jeden, gdy mam 1 przekaźnik) normalnie by był (zgodnie z konfig > przekaźniki > Status załączenia przekaźnika 1. LED) zgaszony dla przekaźnika wyłączonego, zapalony dla włączonego. Nic to nie zmienia w codziennym użytkowaniu
A w trybie "brak sieci" lub sygnalizacja wejścia w konfigurację - by migały wszystkie jednocześnie, jak do tej pory status led. Przełączenie stanu któregoś z kanałow - spowodowałoby co prawda "miganie w negatywie" danego leda - ale nadal mamy widoczną informacje o braku sieci/cloud. I nie musimy używać żadnego gpio/dodatkowego leda do tego

krycha88, 2022-08-04:
Umożliwiłem wybranie tego samego GPIO dla LED konfiguracyjnego oraz do sterowania stanem przekaźnika. @Hrumque przetestuj czy to w ogólne działa :)
https://github.com/krycha88/GUI-Generic/commit/e844d7bd5fde47d773ca2594c2af23edab1158f6

Hrumque, 2022-08-05:
Da się ustawić to samo GPIO (ustawienie w jednym miejscu nie wyklucza użycia w drugim) - ale nie działa tak jak powinno. Dowolny z powodów, by zgasić leda - gasi go. 
Gdy wyjście jest Off = led nie świeci. Gdy Wyjście ON = led świeci, niezależnie od stanu "statusu" 
Gdy wyjście jest off a status powinien migać - na pinie gpio pojawiają się mikro-szpilki (w zasadzie niewidoczne na samym ledzie, tylko na oscyloskopie to widać) więc pewnie w kodzie gdzieś coś ustawia GPIO a zaraz obok go kasuje inna funkcja....

Bo chyba zmieniłeś tylko "zezwolenie na ustawienie" w konfiguratorze, a nie ma zmiany w samym sterowaniu portu GPIO - brakuje "zmień stan na przeciwny" zamiast na sztywno ustawiania 0 lub 1, tam gdzie są hardware porty ruszane w sofcie właściwym 

Mam inny pomysł, choć może wymagać bardziej zaawansowanego przepisania kodu: osobny konfigurator i "HAL" interfejsu "ledy i kontrolki" 
Przykładowo w konfiguratorze modułu przycisk "konfiguracja LEDów:"
**Liczba ledów n** (jak liczba przekaźników itp)
i dla każdego ustawionego leda:
**LEDn : wybór GPIO z listy** (lub nawet do zrobienia w przyszłości - oprócz gpio też  przez expandery i2c, nr leda w szeregu WS2812, zależnie co mamy działające zrobione)
(dla nadgorliwych: pole-string "nazwa leda" np "LED-red" by operować na nazwach później, a nie numerach LED1)

A potem, we wszystkich miejscach gdzie konfigurujemy przypisanie leda (w konfiguracja : led konfig, w przekaźnikach "led załączenia przekaźnika", i gdzie jeszcze kiedyś się doda obsługę leda) - wybieramy tylko z listy LED1, LED2 itd (lub wg przyznanych wcześniej nazw)

Dla każdego leda/kontrolki jedna zmienna (flaga), opisująca jego stan - nie będzie ich przecież nigdy setek, tylko kilka ledów, to nie jest wielkie marnotrastwo zasobów. Wtedy wystarczy binarnie - każda funkcja, która może zmieniać stany led-a zmienia tylko swój jeden bit w statucie przyporzadkowanego jej LEDa: 
Przykładowo:
`przekaźnik1 on > LED1 = LED1 |= 0b0000 0001  ; zaznaczenie chęci zapalenia leda przez ustawienie 1 bez ruszania pozostałych bitów`
`przekaźnik1 off > LED1 = LED1 &= 0b1111 1110  ; zaznaczenie chęci zgaszenia leda przez skasowanie 1 bez ruszania pozostałych bitów`
`status i miganie wolne  > > LED1 = LED1 |= 0b0000 0010 ;`
`status i miganie szybkie > > LED1 = LED1 |= 0b0000 0100 ;  `
`status off  > > LED1 = LED1 &= 0b1111 1001 ;`
itd
A potem w pętelce co jakiś czas sprawdzanie wszystkich statusów ledów i sterowanie odpowiednio przyporządkowanymi im GPIO czy czymtam:
if LED1 == 0 {zgaś leda, bo nikt go nie chciał zapalać}
if inne bity - włącz leda.
if bity odpowiedzialne za miganie - włącz timer od migania i zmień stan itd. 

łatwo byłoby tu przepisywać tylko sam fragment "wypuszczania statusu na hardware" dla danych ledów, w zależności jaki hardware sterujący ledami itp się w przyszłości użyje i wybierze w konfiguratorze, choćby ledy statusów na expanderze i/o i2c, ledy w postaci ikonki na wyświetlaczu, albo ledy w postaci jednego szeregu WS2812b (dowolna liczba ledów na jedym pinie), czy np sygnalizowanie temperatury czy poboru prądu - w postaci drabinki ledów, lub zmiany kolorów...

krycha88, 2022-08-05:
Przetestuj aktualna wersję, poprawiłem to co opisałeś.
https://github.com/krycha88/GUI-Generic/commit/ae257cd49b86cf321d2a51737b2794edfdbdd5fa

Hrumque, 2022-08-05:
Teraz działa zgodnie z przewidywaniem

closed 2022-08-05 by krycha88

Hrumque, 2022-11-13:
Na 5.1.3 niestety przestało działać. 
Gdy chcieć ustawić gpio dla Przekaźnik > LED - wyskakuje przy próbie zapisu "błędne dane" i nie zapisuje, jeśli led jest już użyty wcześniej do konfigu w jakiś sposób.