krycha88/GUI-Generic #68: Czas - do np wyświetlania. NTP

Hrumque, 2022-08-06:
Może warto by dorzucić obsługę czasu (lokalnie na device esp) - nie wygląda to strasznie po stronie wdrożenia:
https://randomnerdtutorials.com/esp8266-nodemcu-date-time-ntp-client-server-arduino/
w konfiguratorze buildera opcja "NTP - obsługa czasu lokalnie"
w konfiguratorze www urządzenia, w innych, "NTP: serwer" do wpisania, NTP: strefa czasowa (dla PL będzie +2), czy używać DST (czas letni/zimowy, jeśli tak to gdy ustawiona flaga w zwracanym tm_isdst to ofset czasu +1h, albo -1h - sprawdzić). 

( Nie wiem jak sprawdzać na ESP łatwo parametry dodatkowe z DHCP - konkretnie Option042 - http://www.faqs.org/rfcs/rfc1769.html - bo serwer dhcp może adres NTP również rozgłaszać, to by z automatu mogło się ustawiać, zamiast na stałe z palca) 

Wtedy w opcjach LCD można by dodawać ekran "czas" (o dacie już pominę). oraz można by warunkować przekaźnik czasem - taki prosty samobieżny programator czasowy ;)

krycha88, 2022-08-07:
Od tego masz harmonogram w cloud - zgadza się, że nie chodzi gdy brak neta ale może kiedyś będzie działać to inaczej.

Pomysł z wyświetlaniem czasu do rozważenia.

krycha88, 2022-08-07:
Dodałem wsparcie wyświetlania czasu na OLED, inne propozycje nie przeszły :)
https://github.com/krycha88/GUI-Generic/commit/2256cd7c66ad7546593c09c72cafd3ed6a237743

closed 2022-08-07 by krycha88