krycha88/GUI-Generic #78: Dodać obsługę Wake on LAN :-)

arturniescior, 2022-08-15:


Hrumque, 2022-08-15:
Ale co konkretnie w temacie WakeOnLan?
Możliwość wysyłania MagicPacket po stronie supla-device ? (po co?)
Możliwość budzenia ESP przez WakeOnLan (przez wifi) - jeszcze bardziej bez sensu, jak esp idzie w deepsleep to żadne wakeonlan go nie obudzi, a jak nie jest deepsleep - to co ma WakeOnLan zrobić? Odebranie WakeOnLan na adres Mac naszego ESP - ma udawać Wyzwalacz akcji, robić za przycisk?

arturniescior, 2022-08-16:
Chodziło mi o odpalanie np komputera za pomocą WoL

djgprv, 2022-08-23:
W routerze nie masz takiej opcji?

djgprv, 2022-09-15:
Wake On LAN Library (ESP8266 & ESP32) - https://github.com/a7md0/WakeOnLan
MIT License

krycha88, 2022-09-16:
Zostało zrealizowane
https://github.com/krycha88/GUI-Generic/commit/087e7c152beafe3e7172a59d5c0fc2b4679ec9f1

closed 2022-09-16 by krycha88