krycha88/GUI-Generic #84: Konfiguracja krańcówki "limit switch" jak przyciski "buttons"

Hrumque, 2022-08-22:
Porównując konfigurację wejścia dla "krańcówki"
https://github.com/krycha88/GUI-Generic/blob/master/src/SuplaWebPageLimitSwitch.h
z konfiguracja wejścia dla przycisku/przełącznika:
https://github.com/krycha88/GUI-Generic/blob/master/src/SuplaWebPageControl.h
widać, że krańcówka jest mocno zubożona. 

Przede wszystkim brakuje przy krańcówce "odwróconej logiki" (w zależności czy krańcówka jest NC czy NO, lub jak jest mechanicznie zamontowana) 
`#define INPUT_LIMIT_SWITCH_INVERSED  "ilsi"` (i wszystkie z tym związane sprawy które odbywają się w SuplaConfigMaganer i później w cześci hardware-wykonawczej)

krycha88, 2022-08-22:
Ograniczenie wynika tylko i wyłącznie po stronie biblioteki:
https://github.com/SUPLA/supla-device/blob/main/src/supla/sensor/binary.cpp

pewnie nie zostało to dorobione, bo można odwrócić logikę po stronie clouda.

Hrumque, 2022-08-22:
Rozumiem i już widzę. Choć uważam, że bardziej przyzwoicie byłoby trzymać się jednej nomenklatury konfiguracji wejść (niezależnie do czego to są wejścia) niż jedne konfigurować tu, drugie tam, a trzecie w ogóle "domyśl się" ;)